SRA-VJTI/sra-board-component #59: Doxygen autogenerator action doesn't seem to work

VedantParanjape, 2022-03-11:
Checking the gh-pages branch, it doesn't seem like the docs are being pushed to it.

SAtacker, 2022-03-11:
> Checking the gh-pages branch, it doesn't seem like the docs are being pushed to it. Also, it doesn't seem updated on the website.
> 

Can you manually point out any difference between the deployment and the change

VedantParanjape, 2022-03-11:
> > Checking the gh-pages branch, it doesn't seem like the docs are being pushed to it. Also, it doesn't seem updated on the website.
> 
> Can you manually point out any difference between the deployment and the change

https://sravjti.in/sra-board-component/files.html no oled.h mpu6050.h

VedantParanjape, 2022-03-11:
> The deployment is active. You can check the logs of workflow. [image]

This doesn't indicate if it does what it is expected to do. For the case, you could just do an echo "hello world" and it will still show action passed.

VedantParanjape, 2022-03-11:
@SAtacker Check the gh-pages branch as well, it is empty. I think you followed this blog: https://github.com/SRA-VJTI/sra-board-component/blob/main/.github/workflows/doxy_deploy.yaml

Some of the args like deploy branch are missing.

SAtacker, 2022-03-11:
That's correct. I'll fix it. Seems like :cow: tam's reviews and my PRs are unreliable.

VedantParanjape, 2022-03-11:
What is this @gautam-dev-maker 🤣

SAtacker, 2022-03-11:
Also why do we have the https://github.com/SRA-VJTI/sra-board-component/tree/main/docs folder? Should I remove it?

closed 2022-03-11 by VedantParanjape