SRA-VJTI/sra-board-component #67: Move from u8g2 library to lvgl library for OLED display.

gautam-dev-maker, 2022-07-02:
lvgl library seems more industry standard than u8g2 library. So we should rewrite all our OLED wrappers in lvgl library.

gautam-dev-maker, 2022-08-25:
@Jamm02 @ChinmayLonkar @Sidshx has anyone been assigned to take up this task? or should I assign one for this?

Sidshx, 2022-08-28:
Picked up the task. @Premraj02 will be accompanying me.

gautam-dev-maker, 2022-09-14:
@Sidshx any updates on the issue?

Sidshx, 2022-09-15:
Started referring resources on u8g2 and lvgl library and getting acquainted with the Wall-E codebase. Will be sharing an implemented work update soon.

Jamm02, 2022-09-21:
@Sidshx updates?

Sidshx, 2022-09-25:
> @Sidshx updates?

Imported and implemented the lvgl library, along with lvgl driver support for the esp-32. Configuration and hardware testing to be done at SRA tomorrow. Working with @Ayush-Kaura

Jamm02, 2022-09-27:
Share results of application of the firmware

Ayush-Kaura, 2022-09-27:
[image]

Basic Hello World Code is flashed from [examples of lvgl on esp32](https://github.com/lvgl/lv_port_esp32)

Ayush-Kaura, 2022-09-30:
[image] 
Bar for LSA Reading

Ayush-Kaura, 2022-10-18:
[image]

Sample Image Display

gautam-dev-maker, 2022-10-18:
the logo can still be made a bit smaller to fit in OLED

gautam-dev-maker, 2022-10-18:
@Ayush-Kaura With next update also put the link to your forked repository which contains the required code for review.

Ayush-Kaura, 2022-10-27:
[image]
Using LVGL v8.3, master branch had some api changes
Files Modified: lv_conf.h for setting lvgl related options inside LVGL folder, lvgl_helpers.c driver for ssd1306, oled.c and main.c inside src, CMakeLists
Link will open draft pr after some of below work done: https://github.com/Ayush-Kaura/sra-board-component/tree/lvgl_port

To-Do
- [ ] lvgl requires functions like [lv_task_handler](https://docs.lvgl.io/8.3/porting/timer-handler.html), [lv_tick_inc](https://docs.lvgl.io/8.3/porting/tick.html) 
- [ ] overall functions structure in oled.c  (function parameters to be passed)
- [ ] i2cdevlib for mpu and lvgl

VedantParanjape, 2022-10-27:
> [image]
> Using LVGL v8.3, master branch had some api changes
> Files Modified: lv_conf.h for setting lvgl related options inside LVGL folder, lvgl_helpers.c driver for ssd1306, oled.c and main.c inside [...]

We are already using i2devlib.

Ayush-Kaura, 2022-10-27:
> We are already using i2devlib. Just check the source code. :)

Sorry I meant using the same for lvgl driver as done for u8g2 previously, so that we can use both mpu and oled together.

> @Ayush-Kaura I don't see the code used to display all these things in the fork.

Added, https://github.com/Ayush-Kaura/sra-board-component/blob/lvgl_port/src/oled.c

VedantParanjape, 2022-10-27:
> Sorry I meant using the same for lvgl driver as done for u8g2 previously, so that we can use both mpu and oled together.
>
>
> Added, [...]

Also include the updated examples/oled source.

Ayush-Kaura, 2022-10-29:
> Also include the updated examples/oled source.

https://github.com/Ayush-Kaura/sra-board-component/blob/lvgl_port/examples/oled/main/main.c

https://user-images.githubusercontent.com/74896007/198845948-6b17e698-d57e-4b10-b9dc-6bd1905872fe.mp4

gautam-dev-maker, 2022-11-05:
As discussed do the following changes: 
* Use `GLOB` in CMakeLists.txt to add source files of LVGL instead of manually adding name of all files
* use `i2c_dev_lib` and test with MPU6050 and OLED both together