SRA-VJTI/sra-board-component #88: Motor Driver rewrite

SuperChamp234, 2023-10-21:
The MCPWM driver has been redesigned, and the legacy driver has been depreciated. 

The new MCPWM driver is object-oriented, where most of the MCPWM submodule has a driver object associated with it. The driver object is created by factory function. IO control function always needs an object handle, in the first place.

The component currently uses the legacy drivers, which needs to be updated.

More info: https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32/migration-guides/release-5.x/5.0/peripherals.html#adc-oneshot-continuous-mode-drivers

VedantParanjape, 2023-10-21:
Object oriented driver in a C code base ?

SuperChamp234, 2023-10-26:
> Object oriented driver in a C code base ?

Yes, we must create interfaces for the MCPWM timer, comparators, generators and operator ourselves which were previously abstracted by idf.