SRA-VJTI/sra-board-component #97: Update LVGL from v8.3 to v9

5iri, 2025-04-21:
We need to update the LVGL graphics library used in `sra-board-component` from version **v8.3** to the latest stable **v9** release. This will ensure long-term support, access to new features, performance improvements, and better compatibility with modern display drivers.

**Tasks:**

- [ ] Review LVGL v9 release notes and migration guide  
- [ ] Identify breaking changes from v8.3 to v9  
- [ ] Update `lvgl` submodule or dependency in the project  
- [ ] Modify existing UI code to comply with LVGL v9 API  
- [ ] Test all UI components for stability and performance  
- [ ] Document any major changes in the implementation

**Resources:**

- LVGL v9 Migration Guide: https://docs.lvgl.io/master/index.html  
- GitHub: https://github.com/lvgl/lvgl

shauryarane05, 2025-04-22:
Assign me this issue 
I will work on this

crisiumnih, 2025-10-16:
@shauryarane05 update??